tscircuit/circuit-to-svg #20: smtpads not rendering properly

seveibar, 2024-08-21:
The smtpads don't render in the following example

```json
[
  {
    type: "source_port",
    source_port_id: "source_port_0",
    name: "pin1",
    pin_number: 1,
    port_hints: [ "-", "pin1", "1" ],
    source_component_id: "source_component_0",
  }, {
    type: "source_port",
    source_port_id: "source_port_1",
    name: "pin2",
    pin_number: 2,
    port_hints: [ "+", "pin2", "2" ],
    source_component_id: "source_component_0",
  }, {
    type: "source_component",
    source_component_id: "source_component_0",
    ftype: "simple_resistor",
    name: "R1",
    manufacturer_part_number: undefined,
    supplier_part_numbers: undefined,
  }, {
    type: "schematic_component",
    schematic_component_id: "schematic_component_0",
    center: {
      x: 0,
      y: 0,
    },
    rotation: 0,
    size: {
      width: 1,
      height: 0.55,
    },
    source_component_id: "source_component_0",
    symbol_name: "boxresistor_horz",
  }, {
    type: "pcb_component",
    pcb_component_id: "pcb_component_0",
    center: {
      x: 0,
      y: 0,
    },
    width: 0,
    height: 0,
    layer: "top",
    rotation: 0,
    source_component_id: "source_component_0",
  }, {
    type: "pcb_board",
    pcb_board_id: "pcb_board_0",
    center: {
      x: 0,
      y: 0,
    },
    width: 10,
    height: 10,
  }, {
    type: "pcb_port",
    pcb_port_id: "pcb_port_0",
    pcb_component_id: "pcb_component_0",
    layers: [ "top" ],
    x: -0.5,
    y: 0,
    source_port_id: "source_component_0",
  }, {
    type: "pcb_port",
    pcb_port_id: "pcb_port_1",
    pcb_component_id: "pcb_component_0",
    layers: [ "top" ],
    x: 0.5,
    y: 0,
    source_port_id: "source_component_0",
  }, {
    type: "pcb_smtpad",
    pcb_smtpad_id: "pcb_smtpad_0",
    pcb_component_id: "pcb_component_0",
    pcb_port_id: "pcb_port_0",
    layer: "top",
    shape: "rect",
    port_hints: [ "1", "left" ],
    x: -0.5,
    y: 0,
  }, {
    type: "pcb_smtpad",
    pcb_smtpad_id: "pcb_smtpad_1",
    pcb_component_id: "pcb_component_0",
    pcb_port_id: "pcb_port_1",
    layer: "top",
    shape: "rect",
    port_hints: [ "2", "right" ],
    x: 0.5,
    y: 0,
  }
]
```

imrishabh18, 2024-08-21:
No worries! Found it

imrishabh18, 2024-08-21:
@seveibar The reason why they are not rendering the `smtpads` is because the `width` and `height` is missing in the json.

seveibar, 2024-08-21:
@imrishabh18 🤦 thanks!!!!!

closed 2024-08-21 by seveibar