tscircuit/circuit-to-svg #42: Silkscreen should be on layer above traces

seveibar, 2024-09-02:
Should have mentioned this earlier, but the silkscreen is technically the topmost layer on a PCB, so it looks a bit strange when the traces are above it

imrishabh18, 2024-09-02:
@seveibar The traces on the pcb-viewer are also on top of the silkscreen https://debug.tscircuit.com/soup_group/core_example1#%7B%22selected_engine%22%3A%22pcb%22%2C%22selected_layout_index%22%3A0%7D . So, I thought that is the correct way

seveibar, 2024-09-02:
Ahh i see, its wrong there too!

seveibar, 2024-09-02:
@imrishabh18 actually for the pcb viewer because you're editing traces it might actually be okay

seveibar, 2024-09-02:
It should probably be a flag or prop for the pcb-viewer, just so that you can see traces youre editing

closed 2024-09-02 by imrishabh18