tscircuit/circuit-to-svg #48: `Infinity` rendered in svg

seveibar, 2024-09-06:
[image]

Soup that reproduces:

```json
[
  {
    "type": "source_port",
    "source_port_id": "source_port_0",
    "name": "pin1",
    "pin_number": 1,
    "port_hints": [
      "pin1",
      "1"
    ],
    "source_component_id": "source_component_0"
  },
  {
    "type": "source_port",
    "source_port_id": "source_port_1",
    "name": "pin2",
    "pin_number": 2,
    "port_hints": [
      "pin2",
      "2"
    ],
    "source_component_id": "source_component_0"
  },
  {
    "type": "source_component",
    "source_component_id": "source_component_0",
    "ftype": "simple_chip",
    "name": "U1"
  },
  {
    "type": "schematic_component",
    "schematic_component_id": "schematic_component_0",
    "center": {
      "x": 0,
      "y": 0
    },
    "rotation": 0,
    "size": {
      "width": 0.4,
      "height": 0.4
    },
    "pin_spacing": 0.2,
    "source_component_id": "source_component_0"
  },
  {
    "type": "schematic_port",
    "schematic_port_id": "schematic_port_0",
    "schematic_component_id": "schematic_component_0",
    "center": {
      "x": 0,
      "y": 0
    },
    "source_port_id": "source_port_0",
    "facing_direction": "up"
  },
  {
    "type": "schematic_port",
    "schematic_port_id": "schematic_port_1",
    "schematic_component_id": "schematic_component_0",
    "center": {
      "x": 0,
      "y": 0
    },
    "source_port_id": "source_port_1",
    "facing_direction": "up"
  },
  {
    "type": "pcb_component",
    "pcb_component_id": "pcb_component_0",
    "center": {
      "x": 0,
      "y": 0
    },
    "width": null,
    "height": null,
    "layer": "top",
    "rotation": 0,
    "source_component_id": "source_component_0"
  },
  {
    "type": "pcb_board",
    "pcb_board_id": "pcb_board_0",
    "center": {
      "x": 0,
      "y": 0
    },
    "width": 10,
    "height": 10
  },
  {
    "type": "pcb_smtpad",
    "pcb_smtpad_id": "pcb_smtpad_0",
    "pcb_component_id": null,
    "pcb_port_id": null,
    "layer": "top",
    "shape": "rect",
    "width": 1,
    "height": 1,
    "port_hints": [
      "pin1"
    ],
    "x": -2.5,
    "y": 1.25
  },
  {
    "type": "pcb_smtpad",
    "pcb_smtpad_id": "pcb_smtpad_1",
    "pcb_component_id": null,
    "pcb_port_id": null,
    "layer": "top",
    "shape": "rect",
    "width": 1,
    "height": 1,
    "port_hints": [
      "pin2"
    ],
    "x": 2.5,
    "y": 1.25
  },
  {
    "type": "pcb_hole",
    "pcb_hole_id": "pcb_hole_0",
    "hole_shape": "round",
    "hole_diameter": 1,
    "x": 0,
    "y": -1.25
  },
  {
    "type": "pcb_port",
    "pcb_port_id": "pcb_port_0",
    "pcb_component_id": "pcb_component_0",
    "layers": [
      "top"
    ],
    "x": -2.5,
    "y": 1.25,
    "source_port_id": "source_port_0"
  },
  {
    "type": "pcb_port",
    "pcb_port_id": "pcb_port_1",
    "pcb_component_id": "pcb_component_0",
    "layers": [
      "top"
    ],
    "x": 2.5,
    "y": 1.25,
    "source_port_id": "source_port_1"
  }
]
````

jarek-foksa, 2024-09-13:
I would replace Infinity with 8388608 (256 x 256 x 256 / 2). In theory it could be a much [larger number](https://svgwg.org/svg2-draft/types.html#Precision), but in practice browsers such as Chrome don't handle properly attributes with huge numeric values.

imrishabh18, 2024-09-16:
This is fixed now, tested in the `main` and I don't see the Infinity issue in svg

closed 2024-09-16 by imrishabh18