tscircuit/circuit-to-svg #52: If an option is enabled `shouldDrawErrors` draw trace_errors

seveibar, 2024-09-09:
[tscircuit/soup zod export for trace_error](https://github.com/tscircuit/circuit-json/blob/main/src/pcb/pcb_trace_error.ts)

```ts
import { point } from "src/common"
import { z } from "zod"

export const pcb_trace_error = z
  .object({
    pcb_error_id: z.string(),
    type: z.literal("pcb_error"),
    error_type: z.literal("pcb_trace_error"),
    message: z.string(),
    center: point.optional(),
    pcb_trace_id: z.string(),
    source_trace_id: z.string(),
    pcb_component_ids: z.array(z.string()),
    pcb_port_ids: z.array(z.string()),
  })
  .describe("Defines a trace error on the PCB")

export type PCBTraceErrorInput = z.input<typeof pcb_trace_error>
export type PCBTraceError = z.infer<typeof pcb_trace_error>
```

seveibar, 2024-09-09:
Needed to fix this issue easily in tests: https://github.com/tscircuit/tscircuit/issues/330#issuecomment-2338382505

seveibar, 2024-11-08:
/bounty $40

A pcb error should be shown as a dotted red line with text indicating the error. The PCB Viewer has an example of this being here: https://pcb-viewer.vercel.app/?path=/story/traceerror--trace-error-circuit-4

However in circuit-to-svg the red text should always be visible, not just on hover

ShiboSoftwareDev, 2024-11-09:
[image]
now need to make this optional

@seveibar an error should only be shown when the pcb_trace_error has an array of two ports in pcb_port_ids so that it has two points to connect am I right?

ShiboSoftwareDev, 2024-11-09:
closed in #114

closed 2024-11-09 by ShiboSoftwareDev