tscircuit/circuit-to-svg #122: Fix left and right side of resistor port being drawn differently

seveibar, 2024-11-13:
Likely relevant file: https://github.com/tscircuit/circuit-to-svg/blob/main/lib/sch/svg-object-fns/create-svg-objects-from-sch-component-with-symbol.ts#L201-L202

[image]

seveibar, 2024-11-13:
/bounty $5

AkashJana18, 2024-11-16:
How do i test it?
I couldn't find the contribution guide either to check how can i test it
Can you please guide @seveibar

imrishabh18, 2024-11-17:
@AkashJana18 To test the specific file use this command `bun test <path-of-test-file>`. This will create a snapshot of the file in the `__snapshots__` directory with the file name `.snap.svg`. And if there is some difference in the image it will create a `.diff.png`.

PR #177 (2025-02-24): update snapshot for resistor
fixes #122