tscircuit/circuit-to-svg #125: Schematic symbols are flipped vertically when no rotation is there

imrishabh18, 2024-11-14:
Test file - https://github.com/tscircuit/core/blob/main/tests/components/normal-components/potentiometer.test.tsx
[image]

Issues to fix
- schRotation 180 and 0 are same
- schematic-symbols and schRotaion 0, if you compare is vertically flipped which should not be the case

[image]

imrishabh18, 2024-11-14:
/bounty $5

anas-sarkez, 2024-11-14:
@imrishabh18 
I'm already Implemented functions might help you 
-flipSymbolOverXAxis
-flipSymbolOverYAxis
in Schematic symbols repo

imrishabh18, 2024-11-14:
@anas-sarkez This issue was regarding finding out why the `schematic-symbol` is flipped over X axis in `circuit-to-svg` compared to how it's shown in the.

Cause if the user goes to the `schematic-symbols` directory and see's the symbol, then the schematic view should replicate that on usage. Only when the rotation is added to props, then we can probably use the functions which you have implemented.

seveibar, 2024-11-17:
@imrishabh18 its possible they are flipped in schematic symbols. Somebody please check your data before flipping!

imrishabh18, 2024-11-17:
@seveibar Yeah, could be. I was seeing the directory of symbols as source of truth on appearance. So, when I imported that to the circuit to svg, it appeared flipped.
But true, we can take a deeper look to understand.

closed 2025-06-13 by imrishabh18