tscircuit/circuit-to-svg #128: Too much spacing on top pins in schematic view

seveibar, 2024-11-15:
[image]

/bounty $10

ShiboSoftwareDev, 2024-11-16:
@seveibar do you have a reproduction?

AkashJana18, 2024-11-16:
/attempt #128

1. Identify the Cause: Check the logic calculating pin positions and spacing for top pins in the schematic view.
2. Adjust Spacing: Modify the vertical alignment or offset logic to reduce excess spacing while maintaining readability.
3. Test Changes: Render multiple components with varying pin counts to ensure consistent and appropriate spacing.

ShiboSoftwareDev, 2024-11-19:
@seveibar I tried to find a repro but couldn't, do you have a snippet for it by any chance?

ShiboSoftwareDev, 2024-11-19:
[image]