tscircuit/circuit-to-svg #159: trace complexity breaks svg

ShiboSoftwareDev, 2025-01-02:
after reaching this complexity:

[image]

just add one trace and it breaks:

[image]

seveibar, 2025-01-09:
@ShiboSoftwareDev nice catch, i think i know why this is happening, it's a bug in https://github.com/tscircuit/core/blob/main/lib/components/primitive-components/Trace/create-schematic-trace-crossing-segments.ts#L12

/bounty $5

vikashkrdeveloper, 2025-06-28:
/attempt #159

Plan: I’ll patch the trace-to-SVG logic to break overly complex traces into smaller <path> segments. This avoids rendering failures in browsers and editors due to long path strings. I’ll segment the trace commands based on character length or segment count, ensure the SVG output remains visually identical, and test against dense PCB layouts to confirm stability.

closed 2025-10-17 by github-actions[bot]