tscircuit/circuit-to-svg #194: Enable drawing `pcb_trace_error`

seveibar, 2025-03-25:
We want to have an optional mode where `pcb_trace_error`s are drawn by circuit-to-svg. There should be a flag that enables it.

See [pcb trace error](https://github.com/tscircuit/circuit-json?tab=readme-ov-file#pcbtraceerror)

You'll need to find the position of each port in order to properly draw a line.

The PCB Viewer already draws errors as shown below

[image]

Should be fairly straightforward.

We want this feature so that we can use visual snapshot testing with DRC errors

/bounty $12

seveibar, 2025-03-25:
Here is a PR that introduces DRC errors, the circuit json from the introduced test can be used to test

https://github.com/tscircuit/core/pull/716

mach5chaos, 2025-03-25:
/attempt #194

ShiboSoftwareDev, 2025-04-22:
@seveibar when the error concerns a net the line should be drawn between the two nearest ports in the net correct?

Abse2001, 2025-04-24:
/attempt #194

ShiboSoftwareDev, 2025-04-26:
@seveibar is this it? I implemented this 6 months ago

[image]

[image]

ShiboSoftwareDev, 2025-04-26:
@imrishabh18 can you take a look at this?

seveibar, 2025-04-26:
@ShiboSoftwareDev oooh we have this? We should use it for svg snapshots in the checks repo!

closed 2025-04-26 by seveibar